flipperdevices/Flipper-iOS-App #10: Add icons and splash screen

evnik, 2020-09-29:
- [ ] Determine sizes of image assets that required for app icons and splash screen (at least for iPhone device, icons for iPad and Mac might be added later)
- [ ] Request / draw required assets
- [ ] Integrate assets into the project

illabo, 2020-10-01:
Could you please clarify whether you mean launch screen (one shown before application code is executed while animating icon to application's first screen view transition) or a literally splash screen to show for some amount of time after app is launched to have some additional time for startup?
If it is the first one should it show the application logo on solid color or maybe we should leave it for later to make it according to [HIG's recommendation](https://developer.apple.com/design/human-interface-guidelines/ios/visual-design/launch-screen) to "Design a launch screen that’s nearly identical to the first screen of your app."?

evnik, 2020-10-01:
@illabo It's the first one and the reason I'd like to have it now is [this requirement](https://developer.apple.com/news/?id=03042020b)
> Starting April 30, 2020, all apps submitted to the App Store must use an Xcode storyboard to provide the app’s launch screen

Later the date was [extended](https://developer.apple.com/news/?id=03262020b) to June 30, 2020, and I'm not sure what is it now, however I think we can just implement it right away using the Flipper logo and probably some animation.

illabo, 2020-10-01:
I was pretty sure that Apple is about to ditch storyboards in the future and suggest to use `UILaunchScreen` plist key in lieu of `Launchscreen.storyboard` file because of SwiftUI was introduced. Strange that they require storyboard launch screen specifically. However it is a piece of cake to implement.

aquam1ne, 2020-10-02:
Here is the archive with fonts and colors, and a few images. 
https://drive.google.com/file/d/15Kkz9ViWOgluyIdPndfdR_xx-uwemZHk/view?usp=sharing

Feel free to mention me if you need particular graphics.

illabo, 2020-10-03:
What kind of assets should be added also for now?

commit 6dbde4e9 (2020-10-04): Merge pull request #14 from illabo/icons

App icons and iOS launch screen and fonts are integrated. Resolved #10
closed 2020-10-04 by evnik