flipperdevices/Flipper-iOS-App #11: Shortcuts integration

evnik, 2020-09-29:


illabo, 2020-10-03:
Adding shortcuts support requires Intents Extension, which depends on SiriKit capability, which in its turn needs an entitled Development Team registered in the Apple Developer program. 
With this mentioned is there some plan for devteam creation?
Are there someone experienced with Siri Shortcuts among contributors?

theabecaster, 2020-10-03:
@illabo I am currently working on implementing this. I am using my own dev account with a provisional bundle of for the extension, do we have a team dev team yet?

illabo, 2020-10-03:
@theabecaster I expect core team members answering the question regarding dev team. 
May I be a help in shortcuts integration task? I have some free time recently enabling me to contribute. :)

theabecaster, 2020-10-03:
Definitely, what I’m working on right now is a versatile service that will enable us to make apo calls to the Siri Shortcuts app. You currently don’t need an extension for this. The intents extension is what I believe you were referring to is the old way to interface with Siri kit but it’s very limited to pre defined intents and certain types of apps

illabo, 2020-10-03:
@theabecaster What I talking about is Apple recommendation to create Intents extension to share Intents between targets preventing code duplication. As mentioned in [this wwdc18 talk](https://developer.apple.com/wwdc18/211). 
I've never used Siri Shortcuts in my projects and all my stub knowledge is based on dab dab sessions' videos and some clicking around in Xcode. So I absolutely trust anyone who got hands dirt dealing with Siri.

evnik, 2020-10-03:
@zhovner and myself are working on Apple Developer Account setup, expecting to have it next week. Thanks for your patience

closed 2023-08-01 by tonyfreeman