flipperdevices/Flipper-iOS-App #8: Setup CI workflow

evnik, 2020-09-29:
Setup workflow in GitHub actions to compile iOS project for simulator and run Unit Tests

Tapanila, 2020-09-30:
@evnik would you be open to use AppCenter (https://appcenter.ms/) to build, test and distribute beta?

evnik, 2020-09-30:
> would you be open to use AppCenter (https://appcenter.ms/) to build, test and distribute beta?

@Tapanila App Center free tier allows only 240 minutes per month for build and test runs, which won't be sufficient for our needs. GitHub Actions is totally free for public repositories, it's a bit more hustle to setup comparing to App Center, but the provided runners are totally the same.
Builds distribution doesn't make much sense for ad-hoc builds, because we'll need to whitelist all testing devices with the limit of 100. I think we can use TestFlight with the public link for this purpose.
Later we might consider to use App Center for crash reports and analytics (if we'd like to have any).

Tapanila, 2020-09-30:
@evnik okay. I think that's good enough reason. so you would like to have Github action that does the following:

1. Compiles the application with xcode
2. Runs the unit tests and reports status on pull request?

Tapanila, 2020-09-30:
Also my understanding from Github actions pricing on Mac. Is that you get 200 minutes a month for free.
Because Mac build minute multiplier is 10.

https://docs.github.com/en/free-pro-team@latest/github/setting-up-and-managing-billing-and-payments-on-github/about-billing-for-github-actions

evnik, 2020-09-30:
@Tapanila
> would like to have Github action that does the following...

yes, that should be enough to start with. Please note that we have two targets: iOS and macOS. We need to set it up for iOS at least, and compilation might be done for Simulator only.
macOS target requires Big Sur, which currently might be not available as a runner, so we can add it later

> you get 200 minutes a month for free

I think this is for private repositories, public are completely free

closed 2020-10-02 by evnik