flipperdevices/Flipper-iOS-App #9: Setup coverage report for unit tests

evnik, 2020-09-29:


illabo, 2020-10-05:
Codecov is definitely a versatile and well known service. Most marvelous is theirs Sourcegraph browser extension for coloring code inline right on GitHub. But to clarify whether we need to use Codecov or some other tool/service here are some questions. 
- Do we need Sourcegraph to be enabled for this repo?
- Are coverage reports in PR discussion thread all we need from Codecov?
Following are what I have against Codecov:
- It's an external service. 
- Its Coverage report looks off my liking. Personally I'd better prefer something looking like [this](https://github.com/jenkinsci/github-pr-coverage-status-plugin) or [that](https://github.com/jenkinsci/github-coverage-reporter-plugin). (Im not suggesting to bring Jenkins but talking about reports look only)
Alternatively, may I suggest a weird idea of reinventing the wheel? It's possible to make the coverage reports as the last step in Actions workflow. It's also possible to have diffs between pr commits and master branch thanks to the ability to keep jobs' artifacts. 
Pros are:
- We could alternate the look of these reports to any extent. 
- No more external services are needed. 
- It's always fun to code useless tools. 🤓 
Cons are:
- It'd take time to maintain. 
- If (or when) we have to add more functionality it should be (re)implemented rather than configured/enabled. 
Sorry in advance for such a bikeshedding comment. 😶

evnik, 2020-10-05:
> Do we need Sourcegraph to be enabled for this repo?

No, this is not required for now.

> Are coverage reports in PR discussion thread all we need from Codecov?

That's pretty much it.

> Personally I'd better prefer something looking like this or that

This kind of report would be also enough to resolve this issue.

I'm fine with custom solution as well, as long as it is a separate project and it doesn't require maintaining of any service.

commit 3aa5bfa9 (2020-10-27): Merge pull request #19 from illabo/codecoverage

Code coverage check is added. Resolves #9
closed 2020-10-27 by evnik