flipperdevices/Flipper-iOS-App #13: Integrate linter for Swift code

evnik, 2020-10-01:
We need to follow common code style in the project. The best way to do it is to have the style checker or linter.
I expect to use [SwiftLint](https://github.com/realm/SwiftLint), but open to other options.
We can start with default rules and update them as needed. SPM is the preferred way to integrate.

illabo, 2020-10-01:
SwiftLint definitely is the best linter to use IMO. However, are you sure it should be setup via SPM? AFAIK it isn't meant to be set via SPM as a part of project. To use systemwide installation it is only need couple lines of bash in Run Script build phase.
In case of CocoaPods installation it downloads specific version of the linter to project pod's directory rather than including the sources and rebuilding it on every build.

evnik, 2020-10-01:
@illabo  SPM now supports binary dependencies, so I was thinking there is a way to do it similarly to CocoaPods. Please explore that option, if there is no way, we'll need to add CocoaPods integration as well

illabo, 2020-10-01:
I believe it could be done. But it’s too much hassle and confusion to do it this way. Please have a look at podspec of SwiftLint. It just downloads a `portable_swiftlint.zip` file to pods dir, inside is just a precompiled swiftlint itself. Most straightforward way to get a mere tool. 
To use it as Swift PM resource someone have to create a dummy Package and point it to grab same file downloaded with CocoaPods. Then on every build we have to look for the file somewhere in Derived Data directory. Not the best way to go. 
It’s much much better to fix a bash script to first try to run system swiftlint and if it isn’t installed wget or curl download it to project’s root and execute it afterwards.
Just 3–5 lines of bash is much better solution IMO. Swift PM is for packages used from code, not for bringing single binary for build script.

commit f392a96e (2020-10-03): Merge pull request #15 from illabo/linter_setup

SwiftLint is integrated. Fixes #13
closed 2020-10-03 by evnik