flipperdevices/Flipper-iOS-App #23: Simplify custom fonts usage

evnik, 2020-10-24:
We have 2 custom font's added to the project (and might have more in the future):
* `Born2bSportyV2` - regular pixel font
* `HelvetiPixel` - bold pixel font

We need to provide an easy way to use those font's in Swift UI code. Something like
```swift
Text("Hello").font(.regularPixel, .body)
/// OR
Text("Hello").font(.regularPixel(.body))
```
I assume we can use approach similar to [this SO answer](https://stackoverflow.com/a/58791769/2595805) (but for custom fonts).

AmeyaPhadnis-2019H1030012G, 2020-10-24:
@evnik Can I give it a try, I am new to the open-source community and looking forward to opportunities to contribute. Thanks in advance!

illabo, 2020-10-24:
@evnik Are we really should be able passing `Font.TextStyle` to custom fonts? TextStyle when used with custom fonts only binds it to resizing with system font of such a style. Baseline size should be passed to `Font.custom` call anyway.
```swift
extension Font {
    static func regularPixel(style: TextStyle = .body) -> Font? { pixelized(style: style, customStyle: .regular) }

    static func boldPixel(style: TextStyle = .body) -> Font? { pixelized(style: style, customStyle: .bold) }

    static func pixelized(
        style: TextStyle = .body,
        customStyle: CustomFontStyle = .regular,
        size: CGFloat = 24) -> Font {
        .custom(customStyle.rawValue, size: size, relativeTo: style)
    }
}

enum CustomFontStyle: String {
    case regular = "HelvetiPixel"
    case bold = "Born2bSportyV2"
}
```
We can just hardcode `.body` style with all the custom fonts and use just the names of fonts as enum cases like that:
```swift
func customFont(_ customStyle: CustomFontStyle) -> Font {
    .custom(customStyle.rawValue, size: 24, relativeTo: .body)
}
```

evnik, 2020-10-24:
> Are we really should be able passing `Font.TextStyle` to custom fonts?

@illabo we don't need to pass that `.body` directly, we even don't need it to be of the `Font.TextStyle` type (means we might declare our own enum with similar names for cases). I just want to have a similar way to call the font modifier

@AmeyaPhadnis-2019H1030012G sure, feel free to raise your PR when you are ready

closed 2022-03-15 by tonyfreeman