flipperdevices/Flipper-iOS-App #29: Mitigate FlipperZero.xcodeproj conflicts

illabo, 2021-03-24:
It is possible to exclude non human readable Xcode project configuration files from version control. There is a tool named [XcodeGen](https://github.com/yonaskolb/XcodeGen/blob/master/Docs/Usage.md) we could find helpful.
While it is great to lower `xcodeproj` merge conflicts rate, but it adds another tool to run on build phase and yet another config file to take care of.
Let's discuss whether we may need this tool?

taiheng, 2021-06-30:
[tuist](https://tuist.io) is another alternative that can also embed the version of tuist binary that was used to generate the project so that this dependency is already included in the repo and dosen't need to be installed with homebrew. I haven't used either though so cannot comment on features.

illabo, 2021-06-30:
@taiheng could you please compare the pros and cons of the tools mentioned in the issue?
The idea is to exclude .xcodeproj files from version control and regenerate it at every contributor’s machine (and gh actions also). Quickly looked at tuist docs and it seems less mature than XcodeGen on a first glance, however there may be some killer features. 
Also the issue should be assessed by @evnik as a repository maintainer.

taiheng, 2021-07-01:
There was another that I'd heard of, I'll gather some data over the weekend.

PR #32 (2021-08-25): Move Core to SwiftPM package to mitigate FlipperZero.xcodeproj conflicts
SwiftPM integration is officially supported by Apple so we don't need for third-party tools like XcodeGen

Closes #29
closed 2021-08-25 by tonyfreeman