flipperdevices/Flipper-iOS-App #173: Uploading firmware to Flipper stops randomly

krupers, 2023-08-08:
[07_27_00 05-08-2023.txt](https://github.com/flipperdevices/flipperzero-firmware/files/12290105/07_27_00.05-08-2023.txt)
[01_07_32 05-08-2023.txt](https://github.com/flipperdevices/flipperzero-firmware/files/12290106/01_07_32.05-08-2023.txt)
[00_58_17 05-08-2023.txt](https://github.com/flipperdevices/flipperzero-firmware/files/12290107/00_58_17.05-08-2023.txt)
[00_39_12 05-08-2023.txt](https://github.com/flipperdevices/flipperzero-firmware/files/12290109/00_39_12.05-08-2023.txt)
[01_08_25 05-08-2023.txt](https://github.com/flipperdevices/flipperzero-firmware/files/12290110/01_08_25.05-08-2023.txt)
### Describe the bug.

With latest releases update 0.88.0 I am encountering an issue with update. I am using a mobile app and usually it takes 5-10 minutes to update the firmware. With 0.88.0 I am trying to update and it pauses on different moments of percentage progress. I tried to reboot the Flipper OS, the app, abandon Sync, and it always stops at random progress. 

### Reproduction

1. Reboot the Flipper OS
2. Start the Flipper app (iOS)
3. Wait for Flipper to be connected and synced
4. Click on update button to upgrade from 0.87.0 to 0.88.0

### Target

Flipper zero update

### Anything else?

Tried this multiple times and maximum progress bar reached was 43%

skotopes, 2023-08-08:
@tonyfreeman can you take a look?

tonyfreeman, 2023-09-04:
Can't see anything useful, it just loses connection for some reason. 
@krupers have you faced this issue since? We need debug logs from device.

tonyfreeman, 2023-10-24:
This might be related to the UI issue when an update is happening but the progress bar is not responsive. Should be fixed in latest testflight release. Feel free to reopen if it's not.

closed 2023-10-24 by tonyfreeman