flipperdevices/Flipper-iOS-App #175: Incorrect display of app update count label

valentineus, 2023-09-02:
**Steps to reproduce**:
1. Open the application.
2. Navigate to the Hub tab.
3. Open the Apps section.

**Expected behavior**:
The label should correctly display the number of apps that require an update (e.g., "10").

**Actual behavior**:
The label incorrectly displays the number of apps for update if there are more than 9.

**Additional information**:
- Phone model: iPhone 13 (MLA23LL/A)
- OS version: 16.6
- App version: 1.6.5

**Screenshots/video**:
[image]

tonyfreeman, 2023-09-04:
Fixed, thanks!

closed 2023-09-04 by tonyfreeman

valentineus, 2023-09-04:
Awesome, thanks a lot! 🚀