flipperdevices/Flipper-iOS-App #176: Precompiled releases

FalsePhilosopher, 2023-10-19:
Can we get precompiled releases so we can sideload qflipper as an option?

tonyfreeman, 2023-10-24:
Hi! Could you be more specific? What kind of problem you trying to solve?

FalsePhilosopher, 2023-10-25:
> Hi! Could you be more specific? What kind of problem you trying to solve?

Adding a precompiled .ipa to the releases page. I forgot my apple account info on my ios test device and don't feel like getting my info reset or using some annoying cert signing service like test flight. A simple checkra1n, wget link, then apt install *.ipa is all anybody should need to use a piece of FOSS on ios instead of having to use or get around walled gardens and other noise like it. So for those who have a rootable ios devices it would simplify/be faster installing qflipper that way, even better is to make a cydia repo and push qflipper there to make it even easier for people with a rooted ios device.