flipperdevices/Flipper-iOS-App #177: Trigger function from macOS (via shortcuts maybe?)

jondkinney, 2023-10-21:
Hi there,

I'm looking to get some integration with macOS so that I can actually trigger a flipper command via my Stream Deck.

Any thoughts on if this is currently possible or what is needed to make this work? 

I'm assuming as a stop gap I could use GUI scripting, but it'd be more reliable and faster to have it triggered via shortcuts, I imagine.

Thanks!

tonyfreeman, 2023-10-24:
Hey! What kind of commands would you like to trigger? Have you tried "Play Alert" and "Emulate/Send archived item" shortcuts commands?

jondkinney, 2023-10-28:
@tonyfreeman to be honest, I'm not 100% sure what it's called yet as I'm new to FlipperZero, but what I'm trying to achieve is a way to control some lights (that came with an RF remote) via my Stream Deck through the macOS Shortcuts plugin for Stream Deck which would talk to the Flipper iOS app running on my m2 MacBook Pro. So whatever command I'd need to send on the FlipperZero to send those on/off signals to the lights is what I'd need to trigger. Hope that helps!

jondkinney, 2023-10-31:
I just realized that what I'm asking for is already possible. It was just not evident before having the device and trying to make it do what I wanted! I didn't realize the verbiage "Emulate/Send archived item" is what I would need to trigger. Sorry to waste your time, but thank you for the response and the tips. I'm triggering shortcuts via my Stream Deck on macOS by running the iPhone app on the desktop with the m2. Working great!

closed 2023-10-31 by jondkinney