flipperdevices/Flipper-iOS-App #224: Shortcut cannot be used to activate IR Remote buttons

venkr, 2024-06-01:
The shortcut only sends the top-level archived entry, and hence doesn't allow you to specify which button to press. 
As a result, a shortcut that runs an IR remote does nothing.

The-Johnson, 2025-02-16:
I'm having the same issue, I also tried creating a remote with only one button saved inside to test, same result, nothing happens when the shortcut executes.

Salzman-Software, 2025-02-18:
There should be an option to save an IR file in the same way as a subghz or nfc file (with one top-level entry and no buttons) to be played individually or with a shortcut.

FGarridoV, 2025-05-28:
Same issue here! Flipper could be a very powerful device for home automation if this can be solved :)