flipperdevices/Flipper-iOS-App #260: IPAD FULLSCREEN

Killer74-hub, 2024-11-27:
Add an fullscreen for the IPad users.

unnamedd, 2024-11-27:
@Killer74-hub This is the beauty of an open-source project: you can do it yourself without having to wait for it to be implemented by our busy friends from the Flipper team 😀

Killer74-hub, 2024-11-27:
> @Killer74-hub This is the beauty of an open-source project: you can do it yourself without having to wait for it to be implemented by our busy friends from the Flipper team 😀

It’s okay. I can’t do it and I got patience

Killer74-hub, 2024-11-27:
> Well... I have the feeling (but this is just a feeling) that you will have to wait for quite some time until another developer in the community decides to implement it, since the iOS project has existed for quite a few years now. I can see they are working on different things, and if I am not mistaken, there are only two people actively working on this project.
> 
> I've implemented support for myself on my iPad, but never submitted any PR. In the end, when I am using Flipper, it tends to be with my phone, or if I am on the computer, I just use Labs or qFlipper.

Could you sent your iPad app?

unnamedd, 2024-11-27:
I do not have it anymore, but also, I don't think their license allows anyone to redistribute the app outside of their official accounts. Sorry not to be able to help here, mate.

alkozin, 2025-01-20:
Hello mates 👋 Some progress here? I can help